ThexXTURBOXx/clion-embedded-esp32 #4: Download Firmware Popup Stuck

samuelbles07, 2022-12-27:
## Expected Behavior
Download firmware finish and debugging UI opened

## Actual Behavior
Stuck in `Firmware Download` popup

[image]

**Debug Configuration**

[image]

**OpenOCD log** (able to see it only when i wait for a couple minute and cancel popup)

```
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

WARNING: boards/esp32-wrover.cfg is deprecated, and may be removed in a future release.
         If your board is ESP32-WROVER-KIT, use board/esp32-wrover-kit-1.8v.cfg instead.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 34 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 87 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu0.flash' size 4096 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 34 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 87 KB
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu1.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu1.flash' size 4096 KB
** Programming Started **
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Erased 28672 bytes in 245.018 ms
Info : PROF: Data transferred in 82.91 ms @ 337.716 KB/s
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Wrote 28672 bytes in 308.365 ms (data transfer time included)
** Programming Finished in 997 ms **
** Verify Started **
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Flash verified in 67.763 ms 
** Verify OK **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Erased 4096 bytes in 81.012 ms
Info : PROF: Data transferred in 21.819 ms @ 183.326 KB/s
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Wrote 4096 bytes in 218.387 ms (data transfer time included)
** Programming Finished in 524 ms **
** Verify Started **
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Flash verified in 54.847 ms 
** Verify OK **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
** Programming Started **
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Erased 184320 bytes in 1124.21 ms
Info : PROF: Data transferred in 634.958 ms @ 283.483 KB/s
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Wrote 184320 bytes in 905.703 ms (data transfer time included)
** Programming Finished in 2256 ms **
** Verify Started **
Info : [esp32.cpu0] Target halted, PC=0x40092612, debug_reason=00000001
Info : PROF: Flash verified in 146.567 ms 
** Verify OK **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : tcl server disabled
Info : Listening on port 4444 for telnet connections
```

## Steps to Reproduce the Problem

  Just run debug

## Specifications

  - Plugin Version: 0.2.0
  - CLion version: **2022.3.1**
  - OpenOCD version: **Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)**
  - GDB Version: **GNU gdb (crosstool-NG esp-2021r2-patch5) 9.2.90.20200913-git**
  - STM32CubeMX version:
  - Toolchain: esp-idf
  - Platform: esp-idf
  - Hardware configuration: Default // esp32 dev module
  - Debugger: ESP-Prog

samuelbles07, 2022-12-27:
If i set `perform` to `none`, it works fine. Maybe we need to add openocd debug level option to the configuration, so if something happen we able to discover more.

ThexXTURBOXx, 2022-12-27:
Thanks for reporting this issue. I will investigate it some time next year since I will not be in my hometown until then :)

samuelbles07, 2023-02-28:
Hi @ThexXTURBOXx , is there any update on this?

ThexXTURBOXx, 2023-02-28:
@samuelbles07 Sadly no since I cannot reproduce the issue yet. I will let you know when there is some progress on this

jeongyeham, 2023-07-04:
I'm having the same issue too, just like samuelbles said
Stuck in Firmware Download popup
So I clone the project and run gradle debug，stop when it stuck
I got the information as follows:
[image]
[image]

jeongyeham, 2023-07-04:
@ThexXTURBOXx When the value of _**perform**_ isn't set to _**none**_, it seems that the problem really exists.
The program loops endlessly through the function  and  returns the value **_“downloadstatus”_**
[image]

ThexXTURBOXx, 2023-07-04:
Thanks for the information! However, I still don't have the hardware to debug this, sadly

PR #7 (2023-07-06): Fix Pop Stuck
Fixes #4 
# WHY
The output of openocd is similar to`** Programming Finished in XXX ms **`
Just as follows:
[image]
[image]
I ran Gradle debug and found it **_never_** ran into function `set(STATUS.FLASH_SUCCESS)`.
The processListener and filter don't find the string correctly, so the future task can't return the value `downloadStatus`.
Openocd in future task is always waiting for the GDB launching,  but the main task is always waiting for the return value and then launches the GDB.
So they're waiting for each other!
**Program Deadlock!!!!**