ThexXTURBOXx/clion-embedded-esp32 #6: File path parameter error from plugin to openocd

jeongyeham, 2023-06-30:
## Expected Behavior
```
D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin\openocd.exe -c "tcl_port disabled" -s D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\share\openocd\scripts -f interface/esp_usb_jtag.cfg -f board/esp32c3-builtin.cfg -c "program_esp C:/Users/38326/Desktop/Hollow_Clock/Firmware_Prj/build/bootloader/bootloader.bin 0x0 verify" -c "program_esp C:/Users/38326/Desktop/Hollow_Clock/Firmware_Prj/build/partition_table/partition-table.bin 0x8000 verify" -c "program_esp C:/Users/38326/Desktop/Hollow_Clock/Firmware_Prj/build/hollow_clock.bin 0x10000 verify" -c "init; reset halt" -c shutdown
 ```

## Actual Behavior
```
D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin\openocd.exe -c "tcl_port disabled" -s D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\share\openocd\scripts -f interface/esp_usb_jtag.cfg -f board/esp32c3-builtin.cfg -c "program_esp D:/Espressif/frameworks/esp-idf/C:/Users/38326/Desktop/Hollow_Clock/Firmware_Prj/build/bootloader/bootloader.bin 0x0 verify" -c "program_esp D:/Espressif/frameworks/esp-idf/C:/Users/38326/Desktop/Hollow_Clock/Firmware_Prj/build/partition_table/partition-table.bin 0x8000 verify" -c "program_esp C:/Users/38326/Desktop/Hollow_Clock/Firmware_Prj/build/hollow_clock.bin 0x10000 verify" -c "init; reset halt" -c shutdown

Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz
Warn : Interface already configured, ignoring
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
** program_esp input args <0x0 verify> **
Info : esp_usb_jtag: serial (60:55:F9:73:61:24)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif
 Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif
 Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
Info : [esp32c3] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 37 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
** Programming Started **
Error: couldn't open D:/Espressif/frameworks/esp-idf/C:/Users/38326/Desktop/Holl
ow_Clock/Firmware_Prj/build/bootloader/bootloader.bin
embedded:startup.tcl:1510: Error: ** Programming Failed **
in procedure 'program_esp'
in procedure 'program_error' called at file "D:\Espressif\tools\openocd-esp32\v0
.12.0-esp32-20230419\openocd-esp32\share\openocd\scripts/target/esp_common.cfg",
 line 190
at file "embedded:startup.tcl", line 1510
```
## Steps to Reproduce the Problem

  1.Just click Run button

## Specifications

  - Plugin Version: 0.2.2
  - CLion version: 2023.2(EAP)
  - OpenOCD version: v0.12.0-esp32-20230419
  - GDB Version: riscv32-esp-elf-gdb-12.1_20221002
  - STM32CubeMX version: NULL
  - Toolchain: riscv32-esp-elf/esp-12.2.0_20230208
  - Toolset: msys64/ucrt64
  - Platform: windows11 pro
  - Hardware configuration: ESP32c3 with builtin USB serial/JTAG

ThexXTURBOXx, 2023-06-30:
This is weird. I will take a look at this as soon as I am home again, but it will not be before Monday, sadly.
Anyway, thanks for reporting this issue already!

jeongyeham, 2023-07-01:
By the way
### I use the batch file “idf_cmd_init.bat” provided in the official installation package to initialize the Toolchain and CMake.
### I didn't do it like the official method provided by JetBrains.
Just only like this:
[image]

The content of the file is as follows:
```
@echo off

:: The script determines location of Git, Python and ESP-IDF.
:: Similar result can be achieved by running export.ps1 from ESP-IDF directory.

:: How the script determines the location of ESP-IDF:
:: 1. try to use the fist input parameter to query configuration managed by idf-env
:: 2. try to use environment variable IDF_PATH to query configuration managed by idf-env
:: 3. try to use local working directory to query configuration managed by idf-env

if "%IDF_TOOLS_PATH%" == "" (
    set IDF_TOOLS_PATH=%~dp0
    echo IDF_TOOLS_PATH not set. Setting to %~dp0
)

if exist "echo" (
    echo "File 'echo' was detected in the current directory. The file can cause problems with 'echo.' in batch scripts."
    echo "Renaming the file to 'echo.old'"
    move "echo" "echo.old"
)

set PATH=%IDF_TOOLS_PATH%;%PATH%
set TEMP_IDF_PYTHON_PATH="%TEMP%\idf-python-path.txt"
set TEMP_IDF_PATH=%TEMP%\idf-path.txt
:: Check whether IDF ID was specified as the first parameter
set PARAM=%1
if /i "%PARAM:~0,7%"=="esp-idf" (
    idf-env config get --property "path" --idf-id %PARAM%>%TEMP_IDF_PATH%
    idf-env config get --property python --idf-id %PARAM%>%TEMP_IDF_PYTHON_PATH%
    set /P IDF_PATH=<"%TEMP_IDF_PATH%"
) else (
    if "%IDF_PATH%" == "" (
        set IDF_PATH=%CD%
    )
    idf-env config get --property python --idf-path %IDF_PATH%\>%TEMP_IDF_PYTHON_PATH%
)

set /P IDF_PYTHON=<%TEMP_IDF_PYTHON_PATH%

set TEMP_IDF_GIT_PATH="%TEMP%\idf-git-path.txt"
idf-env config get --property gitPath>%TEMP_IDF_GIT_PATH%
set /P IDF_GIT=<%TEMP_IDF_GIT_PATH%

set PREFIX=%IDF_PYTHON% %IDF_PATH%
DOSKEY idf.py=%PREFIX%\tools\idf.py $*
DOSKEY esptool.py=%PREFIX%\components\esptool_py\esptool\esptool.py $*
DOSKEY espefuse.py=%PREFIX%\components\esptool_py\esptool\espefuse.py $*
DOSKEY espsecure.py=%PREFIX%\components\esptool_py\esptool\espsecure.py $*
DOSKEY otatool.py=%PREFIX%\components\app_update\otatool.py $*
DOSKEY parttool.py=%PREFIX%\components\partition_table\parttool.py $*

:: Clear PYTHONPATH as it may contain libraries of other Python versions
if not "%PYTHONPATH%"=="" (
    echo Clearing PYTHONPATH, was set to %PYTHONPATH%
    set PYTHONPATH=
)

:: Clear PYTHONHOME as it may contain path to other Python versions which can cause crash of Python using virtualenv
if not "%PYTHONHOME%"=="" (
    echo Clearing PYTHONHOME, was set to %PYTHONHOME%
    set PYTHONHOME=
)

:: Set PYTHONNOUSERSITE to avoid loading of Python packages from AppData\Roaming profile
if "%PYTHONNOUSERSITE%"=="" (
    echo Setting PYTHONNOUSERSITE, was not set
    set PYTHONNOUSERSITE=True
)

:: Get base name of Git and Python
for %%F in (%IDF_PYTHON%) do set IDF_PYTHON_DIR=%%~dpF
for %%F in (%IDF_GIT%) do set IDF_GIT_DIR=%%~dpF

:: Add Python and Git paths to PATH
set "PATH=%IDF_PYTHON_DIR%;%IDF_GIT_DIR%;%PATH%"
echo Using Python in %IDF_PYTHON_DIR%
%IDF_PYTHON% --version
echo Using Git in %IDF_GIT_DIR%
%IDF_GIT% --version

:: Check if this is a recent enough copy of ESP-IDF.
:: If so, use export.bat provided there.
:: Note: no "call", will not return into this batch file.
if exist "%IDF_PATH%\export.bat" %IDF_PATH%\export.bat
```

I have no understanding of JAVA, so I am not sure if these can help you, but I would like to provide more clues to make it easier for you to reproduce the problem and identify the root cause.

ThexXTURBOXx, 2023-07-02:
[image]

Alright, so I finally have taken a closer look at this and I am very sure that this value inside your run configuration is not correct.
Please let me know if that indeed is the case

jeongyeham, 2023-07-03:
My settings are as follows:
[image]

ThexXTURBOXx, 2023-07-03:
Thank you very much, I think I see what's happening. I will fix this issue in a few hours and publish a new version for 2023.2 EAP.
I will let you know when it is done

ThexXTURBOXx, 2023-07-03:
I have pushed an update right now, along with some other fixes.
It should be available soon for 2023.2 EAP (the version you are running).
Thank you very much for reporting this :)

closed 2023-07-03 by ThexXTURBOXx

ThexXTURBOXx, 2023-07-03:
The update has just been approved and released :)
No problem!

commit a9519e25 (2023-07-03): Fix file path parameter

Fixes #6
commit bc00bff8 (2023-07-03): Fix file path parameter

Fixes #6