ThexXTURBOXx/clion-embedded-esp32 #12: com.jetbrains.cidr.execution.debugger.backend.gdb.GDBDriver$GDBCommandException: Remote 'g' packet reply is too long (expected 452 bytes, got 512 bytes)

Yecgaa1, 2023-08-24:
## Expected Behavior
An error occurred while debugging the ESP32S3 and builtin debuggers with your plugin
The program can be burned normally with jtag and your plugin, that's great

[image]
```shell
com.jetbrains.cidr.execution.debugger.backend.gdb.GDBDriver$GDBCommandException: Remote 'g' packet reply is too long (expected 452 bytes, got 512 bytes): 0004004000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000000fefff0c21f0f092325020600000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```
[image]
Thanks

ThexXTURBOXx, 2023-08-26:
I think this is not related to this plugin, but rather a general problem with GDB: https://stackoverflow.com/questions/4896316/gdb-remote-cross-debugging-fails-with-remote-g-packet-reply-is-too-long

hathach, 2023-09-25:
I got into the same issue as well, seem like the "Use GDB" option couldn't detect one shipped with esp-idf. It would be great if we can just specify the path to GDB binary explicitly in case "From Toolchain"'s GDB is not detected. Or maybe I configured it incorectly.

[image]

I am using ubuntu 22.04, maybe it stores toolchain binaries differently from the windows. Here is how it is stored (downloaded and managed espressif install.sh script). I got a few version of everything since I switch between multiple version of IDF (from 4.0, 4.3, 4.4 and 5.0 etc ...)

```
/home/.espressif/tools
├── esp32s2ulp-elf
│   └── 2.28.51-esp-20191205
│       └── esp32s2ulp-elf-binutils
├── esp32ulp-elf
│   ├── 2.28.51-esp-20191205
│   │   └── esp32ulp-elf-binutils
│   └── 2.35_20220830
│       └── esp32ulp-elf
├── esp-rom-elfs
│   └── 20230320
├── openocd-esp32
│   ├── v0.10.0-esp32-20210902
│   │   └── openocd-esp32
│   ├── v0.11.0-esp32-20211220
│   │   └── openocd-esp32
│   ├── v0.11.0-esp32-20220411
│   │   └── openocd-esp32
│   ├── v0.11.0-esp32-20220706
│   │   └── openocd-esp32
│   ├── v0.11.0-esp32-20221026
│   │   └── openocd-esp32
│   └── v0.12.0-esp32-20230419
│       └── openocd-esp32
├── riscv32-esp-elf
│   ├── esp-12.2.0_20230208
│   │   └── riscv32-esp-elf
│   ├── esp-2021r2-8.4.0
│   │   └── riscv32-esp-elf
│   ├── esp-2021r2-patch3-8.4.0
│   │   └── riscv32-esp-elf
│   └── esp-2021r2-patch5-8.4.0
│       └── riscv32-esp-elf
├── riscv32-esp-elf-gdb
│   ├── 11.2_20220823
│   │   └── riscv32-esp-elf-gdb
│   └── 12.1_20221002
│       └── riscv32-esp-elf-gdb
├── xtensa-esp32-elf
│   ├── esp-12.2.0_20230208
│   │   └── xtensa-esp32-elf
│   ├── esp-2021r2-8.4.0
│   │   └── xtensa-esp32-elf
│   ├── esp-2021r2-patch3-8.4.0
│   │   └── xtensa-esp32-elf
│   └── esp-2021r2-patch5-8.4.0
│       └── xtensa-esp32-elf
├── xtensa-esp32s2-elf
│   ├── esp-12.2.0_20230208
│   │   └── xtensa-esp32s2-elf
│   ├── esp-2021r2-8.4.0
│   │   └── xtensa-esp32s2-elf
│   ├── esp-2021r2-patch3-8.4.0
│   │   └── xtensa-esp32s2-elf
│   └── esp-2021r2-patch5-8.4.0
│       └── xtensa-esp32s2-elf
├── xtensa-esp32s3-elf
│   ├── esp-12.2.0_20230208
│   │   └── xtensa-esp32s3-elf
│   ├── esp-2021r2-8.4.0
│   │   └── xtensa-esp32s3-elf
│   ├── esp-2021r2-patch3-8.4.0
│   │   └── xtensa-esp32s3-elf
│   └── esp-2021r2-patch5-8.4.0
│       └── xtensa-esp32s3-elf
└── xtensa-esp-elf-gdb
    ├── 11.2_20220823
    │   └── xtensa-esp-elf-gdb
    └── 12.1_20221002
        └── xtensa-esp-elf-gdb
```

ThexXTURBOXx, 2023-09-25:
@hathach Thanks for the information! I will see if I can add a "Custom" button to the GDB options. Maybe, this will really help :)

hathach, 2023-09-25:
thank you @ThexXTURBOXx for considering to making changes. I notice clion's builtin openocd run/debug also allow to change the GDB executable as "Debugger" in the run/debug instance. I am not sure if that is a good idea to follow, but worth mentioning. It is totally up to you. For now I force the default GDB with idf gdb as workaround ( but need to revert when working with other MCUs).   

[image]

PS: I am testing with kaluga esp32s2 with built-in FTDI JTAG and able to flash/run with your great plugin, but still haven't got step/step debugging working yet. I am new with clion + esp32 openocd, probably need to spend more time on this later on. Though your plugin is the only solution can get me this far, thanks again.

hathach, 2023-09-25:
PS: thinking about this more, I think it probably makes more sense to have Debugger (executable path) option for each run/debug setting like clion's builtin openocd debugger. Since each espressif mcu: esp32, esp32s2, esp32s3, c3 etc.... seems to have different GDB executable. This will make thing easier to work with multiple espressif mcus. Just my observation, it is totally up to you (whichever approach you think best and easiest)  

```
.
├── esp32s2ulp-elf
├── esp32ulp-elf
├── esp-rom-elfs
├── openocd-esp32
├── riscv32-esp-elf
├── riscv32-esp-elf-gdb
├── xtensa-esp32-elf
├── xtensa-esp32s2-elf
├── xtensa-esp32s3-elf
└── xtensa-esp-elf-gdb
```

ThexXTURBOXx, 2023-10-01:
I have pushed an update which should hopefully add a configuration method for the debugger. If it does not work, just let me know!

hathach, 2023-10-03:
thank you for the fix, I have installed 2023.3 EAP to test this out. And it works, the correct GBD executable can be specified for each run configuration with specific chip e.g esp32, esp32s2, esp32s3 etc ...

Though somehow, I still couldn't debug my kaluga (esp32s2) board with built-in debugger (FTDI2232H). But it is probably another issue, I will try to dive into that later on. Thank you for your hard work.

Yecgaa1, 2023-10-09:
I am very glad that this issue has made new progress, and thank you very much for your efforts. @hathach @ThexXTURBOXx

Yecgaa1, 2023-10-09:
With the new version of the plug-in, I have now successfully completed the compilation, burning and debugging ( using Built _ in Debugger ) on ESP32S3, which is my configuration file for reference. @hathach 
[image]

Yecgaa1, 2023-10-09:
However, I found some problems in the new version of the plug-in. Using the Jtag debugger of ESP32 and CH347, the Jtag debugger can be burned normally, but the debugging can not run normally.
[image]

```
com.jetbrains.cidr.execution.debugger.backend.gdb.GDBDriver$GDBCommandException: Remote replied unexpectedly to 'vMustReplyEmpty': PacketSize=4000;qXfer:memory-map:read+;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+
```
[image]

I try to reproduce the specific reasons for this problem in the command line, but it runs normally in the command line, which makes me somewhat confused. Maybe there are some problems in the plug-in part. If you need more information, I 'd be glad to provide @ThexXTURBOXx 
This is the information I run on the command line
```shell
>C:\Users\xtx\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gdb.exe
GNU gdb (crosstool-NG esp-2022r1) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
.gdbinit:3: Error in sourced command file:
"monitor" command not supported by this target.
(gdb) file C:\GitProject\ESP32\hello_world\cmake-build-debug\hello_world.elf
C:GitProjectESP32hello_worldcmake-build-debughello_world.elf: No such file or directory.
(gdb) file C:/GitProject/ESP32/hello_world/cmake-build-debug/hello_world.elf
Reading symbols from C:/GitProject/ESP32/hello_world/cmake-build-debug/hello_world.elf...
(gdb) target remote localhost:6666
Remote debugging using localhost:6666
Remote connection closed
(gdb)
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x40000400 in ?? ()
(gdb) set mem inaccessible-by-default off
(gdb) set remotetimeout 20
(gdb) mon reset halt
JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[esp32.cpu0] requesting target halt and executing a soft reset
[esp32.cpu0] Debug controller was reset.
[esp32.cpu0] Core was reset.
[esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
[esp32.cpu0] Reset cause (3) - (Software core reset)
[esp32.cpu1] requesting target halt and executing a soft reset
[esp32.cpu0] Core was reset.
[esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[esp32.cpu1] Debug controller was reset.
[esp32.cpu1] Core was reset.
[esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
[esp32.cpu0] Reset cause (3) - (Software core reset)
(gdb) flushregs
Register cache flushed.
(gdb) set remote hardware-watchpoint-limit 2
(gdb) thb app_main
Hardware assisted breakpoint 1 at 0x400d536c: file C:/GitProject/ESP32/hello_world/main/hello_world_main.c, line 16.
(gdb) r
The "remote" target does not support "run".  Try "help target" or "continue".
(gdb) c
Continuing.
[esp32.cpu0] Target halted, PC=0x400D536C, debug_reason=00000001
Set GDB target to 'esp32.cpu0'
[esp32.cpu1] Target halted, PC=0x400846F2, debug_reason=00000000
[... 58 lines cut ...]
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : Detected FreeRTOS version: (10.4.3)
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu0] Target halted, PC=0x400D536C, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x400846F2, debug_reason=00000000
Info : Detected FreeRTOS version: (10.4.3)
Info : [esp32.cpu0] Target halted, PC=0x400D5375, debug_reason=00000001
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x400846F2, debug_reason=00000000
```

hathach, 2023-10-09:
@Yecgaa1 oh, that is great to hear it works with s3, I switch to other works, but being able to debug S2/S3 would be great for my work with tinyusb. I will definitely check this out later on.

ThexXTURBOXx, 2023-10-09:
Thanks for the further information! However, I find it quite odd that this error still persists. As per [documentation](http://web.archive.org/web/20230906090518/https://www.sourceware.org/gdb/onlinedocs/gdb/Packets.html#index-vMustReplyEmpty-packet), `gdb` should not reply like that to unknown packets. But, I will see whether I can do anything about that

Yecgaa1, 2023-10-09:
My knowledge of GDB is limited, but I still want to tell you some of my new findings. Hope to help you.
First of all, it is not impossible to debug every time. Occasionally, it can be successfully connected but the code is paused in a very weird place, as shown below.
[image]
[image]
Secondly, I found that regardless of the Porform option selected as any option, the plug-in seems to try to burn the program. I verified this by deleting bootloader.bin, then debugging and finding that openocd failed. Configuring and failing to report errors are shown below. You can see that since bootloader.bin doesn 't exist, its configuration is marked red.
[image]

```shell
C:\Users\xtx\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe -c "tcl_port disabled" -s C:\Users\xtx\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts -f C:/GitProject/ESP32/hello_world/ESP32-ch347.cfg -c "program_esp C:/GitProject/ESP32/hello_world/cmake-build-debug/bootloader/bootloader.bin 0x1000 verify" -c "program_esp C:/GitProject/ESP32/hello_world/cmake-build-debug/partition_table/partition-table.bin 0x8000 verify" -c "init; reset run;" -c "echo \"** Program Flash Complete! **\""
[0mOpen On-Chip Debugger v0.12.0-c5dd1ab0 (2023-10-08-22:53)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use
'transport select <transport>'.
** program_esp input args <0x1000 verify> **[0m
Info : clock speed 4000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
, part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
, part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
, part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
, part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Target halted, PC=0x400846F2, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x400846F2, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 34 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 87 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu0.flash' size 4096 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 34 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 87 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu1.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu1.flash' size 4096 KB
** Programming Started **
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Error: couldn't open C:/GitProject/ESP32/hello_world/cmake-build-debug/bootloade
r/bootloader.bin
embedded:startup.tcl:1524: Error: ** Programming Failed **
in procedure 'program_esp'
in procedure 'program_error' called at file "C:\Users\xtx\.espressif\tools\openo
cd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts/target/esp_c
ommon.cfg", line 434
at file "embedded:startup.tcl", line 1524
```

jeongyeham, 2023-10-22:
@Yecgaa1 
Please show the OpenOCD information in Run window, just like below, thanks!
[image]

Yecgaa1, 2023-10-22:
I am sorry now can not provide to you, I am on a business trip, I will reply to you next Wednesday

Please wait a little, thank you.
> @Yecgaa1 [#12 (comment)](https://github.com/ThexXTURBOXx/clion-embedded-esp32/issues/12#issuecomment-1752750400) Please reproduce the above status and show the OpenOCD information in Run window, just like below, thanks! [image]

Yecgaa1, 2023-10-25:
I 'm sorry, but I can 't reproduce this bug, because the previous bug is back.
A sad news.
It happened in I upgrade my ESP-IDF from 5.0.x to 5.1.x, the changes which I see at first, it's that now all esp32 chips with xtensa architecture use the same gdb which in xtensa-esp-elf-gdb folder, but not in esp32-gcc folder, maybe that is the reason why the bug comes again
```shell
com.jetbrains.cidr.execution.debugger.backend.gdb.GDBDriver$GDBCommandException: Remote 'g' packet reply is too long (expected 388 bytes, got 420 bytes): 0004004000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000000feffbcc296fec51c2500060000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```
```shell
C:\Users\xtx\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin\openocd.exe -c "tcl_port disabled" -s C:\Users\xtx\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\share\openocd\scripts -f C:/GitProject/ESP32/hello_world/ESP32-ch347.cfg -c "program_esp C:/GitProject/ESP32/hello_world/cmake-build-debug/bootloader/bootloader.bin 0x1000 verify" -c "program_esp C:/GitProject/ESP32/hello_world/cmake-build-debug/partition_table/partition-table.bin 0x8000 verify" -c "init; reset halt" -c "echo \"** Program Flash Complete! **\""
[0m[0mOpen On-Chip Debugger v0.12.0-44d53e8c (2023-10-08-21:58)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING!!! OpenOCD binary version (Open On-Chip Debugger v0.12.0-44d53e8c (2023-
10-08-21:58)) doesn't match the .cfg file version (v0.12.0-esp32-20230921)
** program_esp input args <0x1000 verify> **
Info : clock speed 4000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
, part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
, part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
, part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
, part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000001
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32.cpu0'
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 36 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu0.flash' size 4096 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 36 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu1.flash' size 4096 KB
Info : Using flash bank 'esp32.cpu1.flash' size 4096 KB
** Programming Started **
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : PROF: Erased 28672 bytes in 1793.08 ms
Info : PROF: Data transferred in 2776.54 ms @ 10.0845 KB/s
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : PROF: Wrote 28672 bytes in 4322.62 ms (data transfer time included)
** Programming Finished in 14477 ms **
** Verify Started **
Info : [esp32.cpu0] Target halted, PC=0x400BE636, debug_reason=00000001
Info : PROF: Flash verified in 1341.91 ms
** Verify OK **
** program_esp input args <0x8000 verify> **
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica)
[... 24 lines cut ...]
** Verify Started **
[... cut ...]
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
** Program Flash Complete! **
Info : tcl server disabled
Info : Listening on port 4444 for telnet connections
```

Yecgaa1, 2023-10-25:
Well, it looks like that problem doesn't exist anymore
Maybe it 's because I upgraded ESP-IDF from 5.0.x to 5.1.x
Anyway, now the only problem is that it will burn every time. But I believe you will fix it soon.
@ThexXTURBOXx 

> I am sorry now can not provide to you, I am on a business trip, I will reply to you next Wednesday
>
> Please wait a little, thank you.

tbe, 2024-01-13:
I still experience the original issue on a linux system: I have defined the gdb from the ESP-IDF toolchain, and i checked, that this GDB works. But for some reason the plugin still seems to use it, even if configured in the run configuration.

Yecgaa1, 2024-04-04:
> I still experience the original issue on a linux system: I have defined the gdb from the ESP-IDF toolchain, and i checked, that this GDB works. But for some reason the plugin still seems to use it, even if configured in the run configuration.

[image]
Have you ever tried this?

PR #13 (2023-10-01): Add debugger configuration
Fixes #12
closed 2024-04-04 by Yecgaa1