ThexXTURBOXx/clion-embedded-esp32 #14: Fix the root path

jeongyeham, 2023-10-22:
Hi, I've updated to the latest version of the plugin, and I'm excited about the new features that have been added, but the value of the variable `root`(whether String or Virtualfile) has been redirected to the root directory of ESP-IDF again in the latest plugin, instead of the current project directory, causing the BIN file in the configuration panel to be displayed as an absolute path again. 
https://github.com/ThexXTURBOXx/clion-embedded-esp32/blob/fe13062220f67b94a46e0ed4cfde31c431adf385/src/main/java/esp32/embedded/clion/openocd/OpenOcdConfigurationEditor.java#L112-L113
https://github.com/ThexXTURBOXx/clion-embedded-esp32/blob/fe13062220f67b94a46e0ed4cfde31c431adf385/src/main/java/esp32/embedded/clion/openocd/OpenOcdConfigurationEditor.java#L167-L168
Below is the debug message for my actual project:
[image]

[image]
I've been wondering if  could use the CMake info in CLion to find the build directory. Is this feasible?

closed 2023-10-28 by jeongyeham

ThexXTURBOXx, 2023-10-28:
Is this issue fixed? As far as I remember I copied most of the behaviour of the built-in OpenOCD runner to mimic how it achieves stuff.
But, I would need to take a close look at it again, if this issue still persists

jeongyeham, 2023-10-28:
Yes, I cleared everything related to IDEA and let gradle reconfigure the project and it works fine!
But a little question about this line:
https://github.com/ThexXTURBOXx/clion-embedded-esp32/blob/fe13062220f67b94a46e0ed4cfde31c431adf385/src/main/java/esp32/embedded/clion/openocd/OpenOcdConfigurationEditor.java#L248
Why call the `getParent()` method at the end? 
Isn't the variable with index `0` in virtualFileList `contentRoots` already the value that should be returned?
I'd appreciate your guidance！

ThexXTURBOXx, 2023-10-28:
See `com.jetbrains.cidr.cpp.embedded.stm32cubemx.ConvertProjectAction`:

```java
@Nullable
public static VirtualFile getProjectDir(@NotNull final Stream<Module> modules) {
    if (modules == null) {
        $$$reportNull$$$0(2);
    }
    return modules.map((Function<? super Module, ?>)ModuleRootManager::getInstance).map((Function<? super Object, ?>)ModuleRootModel::getContentRoots).flatMap((Function<? super Object, ? extends Stream<? extends VirtualFile>>)Stream::of).filter(VirtualFile::isDirectory).findFirst().orElse(null);
}
```

But yes, you are right. The `.getParent()` at the end is wrong and is most likely an artifact of my process of copy-pasting stuff.
I will remove it and release a new version soon!