ThexXTURBOXx/clion-embedded-esp32 #15: Unable to Update Offset

synologic, 2023-11-22:
When trying to edit an existing configuration, the offset is not updatable.

If i change it, the Apply button does not enable and clicking OK does not save the offset value.

It works when creating a new configuration, but not when updating.

ThexXTURBOXx, 2023-11-22:
Not reproducible on my end:

https://github.com/ThexXTURBOXx/clion-embedded-esp32/assets/12763829/d0aedf75-23e3-4f6a-88a7-f1601aaeb5c5

synologic, 2023-11-22:
Hey Nico,

I meant Offset not Bootloader Offset :)

[image]

ThexXTURBOXx, 2023-11-22:
Ah, well, should have checked closer :)
Thanks for reporting back! I have identified and fixed the issue in https://github.com/ThexXTURBOXx/clion-embedded-esp32/commit/83c12816367d0b30653b8f19de313a522fce68af
I have also already uploaded it to the Marketplace. Fix should be published soon!

closed 2023-11-22 by ThexXTURBOXx

synologic, 2023-11-24:
Nico, if it's not too much to ask, 0.3.6 is only available for EAP, can you make it available for the 2023.2 — 2023.2.2 ?
Thanks :)

ThexXTURBOXx, 2023-11-25:
Sorry, I don't like backporting stuff because it breaks my semantic versioning. I would just suggest waiting a bit until 2023.3 is officially available (which won't take too long, I think - it's already in beta) :)