ThexXTURBOXx/clion-embedded-esp32 #18: Add Peripherals Tab is missing

fjwbarnard, 2024-01-24:
## Expected Behavior
With the Embedded GDB Server and OpenOCD Run & Debug configuration templates of at least CLion v2023.3.2 there is a Peripherals tab with which you can load MCU `.svd` files and view the MCU peripheral values when on a break point, e.g.  
[image]

These .svd files are available from https://github.com/espressif/svd

## Actual Behavior
The Peripherals tab is missing in the OpenOCD Run & Debug (ESP32) configuration:
[image]

## Steps to Reproduce the Problem

  1. Setup and run an OpenOCD Run & Debug (ESP32) configuration
  2. View the Debug console and see that there is no Peripherals tab

## Specifications

  - Plugin Version: 0.3.6
  - CLion version: 2023.3.2
  - OpenOCD version: v0.12.0-esp32-20230921 
  - GDB Version: 12.1_20231023 (xtensa-esp-elf-gdb-3.12)
  - STM32CubeMX version: N/A
  - Toolchain: xtensa-esp32s3-elf (esp-12.2.0_20230208)
  - Platform: Windows 11
  - Hardware configuration: ESP32S3 with build-it USB-JTAG

jeongyeham, 2024-09-22:
@ThexXTURBOXx 
Hi, Nico, perhaps we can take a look at this open-source author's [project](https://github.com/yunyizhi/ESP-IDF-for-Clion/releases/tag/0.4.1). He has included debug updates in his latest project release and had a tab for peripherals.

ThexXTURBOXx, 2024-09-22:
The best solution would be to update this entire plugin. I met the guy responsible for the embedded development tools from CLion at embedded world in Nuremberg a few months ago and had a nice talk with him about my fork.
He said that he had planned to integrate functionalities from this plugin into CLion itself at some point, but has not yet gotten around to really do that. Maybe, at some point, he will, though.
Basically, most of the code from this project is based on the intrinsic embedded development tools - just with a few extra tweaks. However, since only a few things got updated since then (basically only compatibility changes), most of the code is still the same as in 2019 (or something like that).
If anyone has enough time to dig deeper into this plugin, he can "copy-paste" most of the code from CLion into this project and apply the modifications that had been made over the years in order for this plugin to be updated properly. This would also include a nice Peripherals tab and many more things!