ThexXTURBOXx/clion-embedded-esp32 #25: Reset command during OpenOCD Debug session throws CLion IDE (v2024.2+) Error

FrancoisB-HEX, 2024-10-08:
Related to https://github.com/ThexXTURBOXx/clion-embedded-esp32/issues/21

## Expected Behavior
1. Click the Reset command button during an OpenOCD JTAG Debug session 
2. JTAG Reset command is executed

## Actual Behavior
1. Click the Reset command button during an OpenOCD JTAG Debug session 
2. IDE Error with exception:
```
com.intellij.openapi.diagnostic.RuntimeExceptionWithAttachments: Access is allowed from Event Dispatch Thread (EDT) only; see https://jb.gg/ij-platform-threading for details
Current thread: Thread[#20515,CidrDebuggerCommandExecutor (CidrDebugProcess#2 'OpenOCD ESP32 app'),4,main] 382617841 (EventQueue.isDispatchThread()=false)
SystemEventQueueThread: Thread[#49,AWT-EventQueue-0,6,main] 869468310
```

## Steps to Reproduce the Problem

  1. Setup and run an OpenOCD JTAG Debug session
  1. Attempt to Reset the board

## Specifications

  - Plugin Version: 0.4.0
  - CLion version: 2024.2.2
  - OpenOCD version: v0.12.0-esp32-20240726
  - GDB Version: 14.2_20240403 esp32s3
  - STM32CubeMX version: N/A
  - Toolchain: ESP-IDF v5.3.1-395-g707d097b01
  - Platform: Windows 11
  - Hardware configuration: ESP32-S3 USB built-in JTAG

ThexXTURBOXx, 2024-10-26:
Thanks for the report! I just wanted to fix this bug, but don't know exactly see where the problem lies. Dou you have any detailed stack trace, i.e., some information on which instruction fails?

peeter123, 2024-11-13:
I have this issue as well when pressing the run button, using CLion v2024.2.2 on Windows 11

The stack trace is as follows:
```
2024-11-13 16:11:48,345 [707180661]   INFO - #c.i.e.r.ExecutionUtil - Error running 'main-networking-firmware.elf':<br>Access is allowed from Event Dispatch Thread (EDT) only; see https://jb.gg/ij-platform-threading for details
Current thread: Thread[#20115,DefaultDispatcher-worker-70,6,main] 959660410 (EventQueue.isDispatchThread()=false)
SystemEventQueueThread: Thread[#68,AWT-EventQueue-0,6,main] 1812059680
com.intellij.openapi.diagnostic.RuntimeExceptionWithAttachments: Access is allowed from Event Dispatch Thread (EDT) only; see https://jb.gg/ij-platform-threading for details
Current thread: Thread[#20115,DefaultDispatcher-worker-70,6,main] 959660410 (EventQueue.isDispatchThread()=false)
SystemEventQueueThread: Thread[#68,AWT-EventQueue-0,6,main] 1812059680
	at com.intellij.util.concurrency.ThreadingAssertions.createThreadAccessException(ThreadingAssertions.java:177)
	at com.intellij.util.concurrency.ThreadingAssertions.throwThreadAccessException(ThreadingAssertions.java:172)
	at com.intellij.util.concurrency.ThreadingAssertions.assertEventDispatchThread(ThreadingAssertions.java:63)
	at com.intellij.ide.impl.DataManagerImpl.getDataContext(DataManagerImpl.java:287)
	at com.intellij.openapi.wm.impl.WindowWatcher.getFocusedWindowForProject(WindowWatcher.java:298)
	at com.intellij.openapi.wm.impl.WindowWatcher.suggestParentWindow(WindowWatcher.java:211)
	at com.intellij.openapi.wm.impl.WindowManagerImpl.suggestParentWindow(WindowManagerImpl.kt:219)
	at com.intellij.openapi.ui.impl.DialogWrapperPeerImpl.<init>(DialogWrapperPeerImpl.java:101)
	at com.intellij.openapi.ui.impl.DialogWrapperPeerFactoryImpl.createPeer(DialogWrapperPeerFactoryImpl.java:21)
	at com.jetbrains.rdserver.ui.dialogs.BackendDialogWrapperPeerFactory.createPeer(BackendDialogWrapperPeerFactory.kt:39)
	at com.intellij.openapi.ui.DialogWrapper.createPeer(DialogWrapper.java:884)
	at com.intellij.openapi.ui.DialogWrapper.<init>(DialogWrapper.java:238)
	at com.intellij.ui.messages.AlertDialog.<init>(AlertMessagesManager.kt:103)
	at com.intellij.ui.messages.AlertMessagesManager.showMessageDialog(AlertMessagesManager.kt:67)
	at com.intellij.ui.messages.MessagesServiceImpl.showMessageDialog(MessagesServiceImpl.java:54)
	at com.jetbrains.rdserver.ui.BackendMessagesService.showMessageDialog(BackendMessagesService.kt:33)
	at com.intellij.openapi.ui.Messages.showDialog(Messages.java:202)
	at com.intellij.openapi.ui.Messages.showDialog(Messages.java:163)
	at esp32.embedded.clion.openocd.Informational.showPluginError(Informational.java:55)
	at esp32.embedded.clion.openocd.OpenOcdLauncher.createProcess(OpenOcdLauncher.java:80)
	at com.jetbrains.cidr.execution.CidrLauncher.startProcess(CidrLauncher.java:25)
	at com.jetbrains.cidr.execution.CidrCommandLineState.lambda$startProcess$1(CidrCommandLineState.java:40)
	at com.jetbrains.cidr.execution.CidrCoroutineHelper$runComputableWithModalProgressBlocking$1$1.invoke(cidrCoroutineHelper.kt:32)
	at com.intellij.openapi.progress.CoroutinesKt.blockingContextInner(coroutines.kt:339)
	at com.intellij.openapi.progress.CoroutinesKt$blockingContext$2.invokeSuspend(coroutines.kt:232)
	at com.intellij.openapi.progress.CoroutinesKt$blockingContext$2.invoke(coroutines.kt)
	at com.intellij.openapi.progress.CoroutinesKt$blockingContext$2.invoke(coroutines.kt)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:62)
	at kotlinx.coroutines.CoroutineScopeKt.coroutineScope(CoroutineScope.kt:261)
	at com.intellij.openapi.progress.CoroutinesKt.blockingContext(coroutines.kt:231)
	at com.jetbrains.cidr.execution.CidrCoroutineHelper$runComputableWithModalProgressBlocking$1.invokeSuspend(cidrCoroutineHelper.kt:31)
	at com.jetbrains.cidr.execution.CidrCoroutineHelper$runComputableWithModalProgressBlocking$1.invoke(cidrCoroutineHelper.kt)
	at com.jetbrains.cidr.execution.CidrCoroutineHelper$runComputableWithModalProgressBlocking$1.invoke(cidrCoroutineHelper.kt)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:62)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:163)
	at kotlinx.coroutines.BuildersKt.withContext(Unknown Source)
	at com.intellij.platform.util.progress.ProgressPipeImpl.collectProgressUpdates(ProgressPipe.kt:43)
	at com.intellij.openapi.progress.impl.PlatformTaskSupport$runWithModalProgressBlockingInternal$2$taskJob$1.invokeSuspend(PlatformTaskSupport.kt:141)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:608)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:873)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:763)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:750)
```

ThexXTURBOXx, 2024-11-13:
@peeter123 Thanks for the feedback! I have pushed version 0.4.1, which should (hopefully) fix this!

closed 2024-11-13 by ThexXTURBOXx

ThexXTURBOXx, 2024-11-15:
Perfect, nice to hear that, thank you very much for providing the required information to fix this! :)

commit 736ea376 (2024-11-13): Fix error messages being shown in wrong thread

Fixes #25